FarmBot/farmbot-arduino-firmware #18: Arduino is going sloowwww

roryaronson, 2015-01-02:
My commands seem to be delayed by 30+ seconds. I tried rebuilding everything (rpi controller first, then Arduino) I was then getting the problems that [these folks](http://forum.arduino.cc/index.php?topic=83079.0) are having, where I can't successfully upload the sketch to the Arduino due to a timeout error. If I try pressing the reset button a few times and immedietely try uploading after, then it sometimes completes, but it still operates very slowly...

By that forum thread, it may be a issue with the bootloader in the Arduino, a clean power source problem, an issue with avrdude, or an FTDI driver issue.

The only other thing I can think of is that maybe the EEPROM got filled up and is causing errors? Does the arduino-controller currently clear the EEPROM on boot? http://arduino.cc/en/Tutorial/EEPROMClear

roryaronson, 2015-01-03:
Hmmm ok, I destroyed everything and reinstalled it all. I was able to upload to the Arduino no problem... (??!)

Now, the web app sends commands really fast, but menu.rb seems to still be delayed by ~20 seconds.

TimEvWw, 2015-01-03:
The only time I have issues uploading is when you have the controller
running and communicating while you try to upload. That will almost always
give timeout.

The web app is event based and created to be fast. As soon as a message
arrives, the command processor is notified. The menu.rb just writes the
command to the database. The command processor will check that database
once every x seconds when no message is received. To make it really fast,
the menu should also become a mesh client.

roryaronson, 2015-01-03:
Ok this makes sense.

Maybe there was a phantom process trying to use the Arduino that I didn't close all the way. Maybe the startup service? Either way it seems to be working now.

menu.rb used to be a lot faster, (only ~1-2 seconds of lag). Did you change something?

closed 2015-09-29 by roryaronson