FarmBot/farmbot-arduino-firmware #20: Movements switch direction inadvertently

roryaronson, 2015-03-10:
There is some weird stuff going on with the moments. Sometimes a command to move in one direction causes it to move a bit, then reverse for no reason, then move forward again. Its weird. Its like the direction pin on the RAMPS board is flipping inadvertently... This has been happening for a while, but I was never sure if it was weird stuff with the pi controller or the meshblu commands, etc. I think it has to do with the Arduino. Any ideas?

TimEvWw, 2015-03-10:
It shouldn't be able to switch direction during the execution of a command.
The direction is written in the beginning of the command. Once that is
done, only the step pin is written.

TimEvWw, 2015-03-11:
The only command the mesh blu can give that interferes with the execution
of a command in arduino is the emergency stop. But that would stop all
movement, not switch direction.

roryaronson, 2015-03-11:
Hmmm. Maybe it has something to do with my driver tuning?

Or, is it possible that the direction pin needs to set periodically or even consistently to avoid 'voltage drift' and flipping?

TimEvWw, 2015-03-11:
I tend more towards the stepper driver itself making a bad connection.
Those bits don't suddenly flip sponanously. It would defeat the whole
purpose of a micro controller. You would only have something like that in
the arduino is damaged or if your power supply is acting up.

TimEvWw, 2015-03-13:
If you want I can add debugging messages for when the pin for movement is
set

roryaronson, 2015-03-14:
I think I just need to tune the drivers and check the connections. It's probably something there. If not, then having that debug message would be nice

closed 2017-02-06 by roryaronson