FarmBot/farmbot-arduino-firmware #32: Z-movements don't work

roryaronson, 2015-09-29:
The connection to my motor is good, the issue is that the Arduino doesn't seem to be sending any steps. This is what happens:

z=0
I send a command to move to z=1000
Pi receives the command
Bot does not move
Pi updates log with current location, which is still z=0

roryaronson, 2015-09-30:
This seems to be an issue with Y movements as well. I think this is what Mat was talking about, @RickCarlino

TimEvWw, 2015-09-30:
The Z axis is default set to negative mode. Z = 0 is the top and it can't
go higher. It should move when you move to -1000.

roryaronson, 2015-09-30:
Oh okay, that's probably an issue because the web app manual control widget doesn't allow negative numbers, right @RickCarlino?

Just found out that my Y motor wasn't working at all lol. Its the one I got from Anaheim Automation... Swapping it out now and I should have X and Y movement.

RickCarlino, 2015-09-30:
Highly likely. I will take a look after finishing up crop removal on the
farm designer.

closed 2017-02-06 by roryaronson