FarmBot/farmbot-arduino-firmware #38: Error while performing ino build

davetom13, 2016-07-24:
I receive the following error when I issue "ino build"

~/working/farmbot-arduino-firmware$ ino build
Searching for Board description file (boards.txt) ... /usr/share/arduino/hardware/arduino/boards.txt
Searching for Arduino lib version file (version.txt) ... /usr/share/arduino/lib/version.txt
Detecting Arduino software version ...  1.0.5 (1:1.0.5+dfsg2-2)
Searching for Arduino core library ... /usr/share/arduino/hardware/arduino/cores/arduino
Searching for Arduino variants directory ... /usr/share/arduino/hardware/arduino/variants
Searching for Arduino standard libraries ... /usr/share/arduino/libraries
Searching for make ... /usr/share/arduino/hardware/tools/avr/bin/make
Searching for avr-gcc ... /usr/share/arduino/hardware/tools/avr/bin/avr-gcc
Searching for avr-g++ ... /usr/share/arduino/hardware/tools/avr/bin/avr-g++
Searching for avr-ar ... /usr/share/arduino/hardware/tools/avr/bin/avr-ar
Searching for avr-objcopy ... /usr/share/arduino/hardware/tools/avr/bin/avr-objcopy
src/src.ino
Searching for Arduino lib version file (version.txt) ... /usr/share/arduino/lib/version.txt
Detecting Arduino software version ...  1.0.5 (1:1.0.5+dfsg2-2)
Scanning dependencies of src
Scanning dependencies of arduino
Scanning dependencies of EEPROM
Scanning dependencies of Servo
src/CurrentState.cpp
src/F41Handler.cpp
src/ParameterList.cpp
src/TimerOne.cpp
src/F44Handler.cpp
src/F82Handler.cpp
src/PinGuardPin.cpp
In file included from src/ParameterList.h:6:0,
                 from src/PinGuardPin.h:17,
                 from src/PinGuardPin.cpp:2:
src/PinGuard.h:35:2: error: ‘PinGuardPin’ does not name a type
  PinGuardPin pinGuardPin[5];
  ^
make: *** [.build/mega2560/src/PinGuardPin.o] Error 1
Make failed with code 2_

gabrielburnworth, 2016-07-24:
Try the stable release: https://github.com/FarmBot/farmbot-arduino-firmware/releases

davetom13, 2016-07-24:
I’ve tried the release version and in the raspberry terminal I get the following msg. and motors do not move. So i thought to try the latest.

 {"params":{"name":"moveRelative","speed":100,"x":-1000},"method":"single_command.MOVE RELATIVE","id":"9a56f8d8-6918-423b-85fd-6864754d102e"}
RPI MSG: G0 X0 Y0 Z0
reporting_end_stops is a valid GCode, but no input handler method exists
reporting_end_stops is a valid GCode, but no input handler method exists

davetom13, 2016-07-24:
Ended up getting the motors to move by removing mechanical endstops and jumpering connections.

closed 2016-07-24 by davetom13

roryaronson, 2016-07-24:
@davetom13 glad you figured this one out. We're going to be using rotary encoders instead of endstops on the production devices, so I think the endstop code is in only a semi functional state right now. We just use jumpers too.

davetom13, 2016-07-24:
@roryaronson How would I speed up the motors? It appears the motors are set at 100 and the only option I see in the arduino config is "const int MOVEMENT_INTERRUPT_SPEED = 100;".

Is the speed set by the web API or can this be hard coded in the arduino config.

TimEvWw, 2016-07-25:
@davetom13 <https://github.com/davetom13> You can change the default speed
for the motors in the config.h, MOVEMENT_MAX_SPD_X_DEFAULT. It loads the
speed settings in EEPROM on first boot. After that, you have to set the
parameters using the write parameter command: F22 Px Vy, where x = 71, 72
or 73 for maximum speed of x, y and z, and the y value is the speed in
steps per second. I haven't checked if those parameters are accessible from
the web page.

roryaronson, 2016-07-25:
Those parameters are accessible from the web app!