FarmBot/farmbot-arduino-firmware #42: X2 Dont work

Daniil145, 2016-12-14:
I use [image]
X2 dont work
only X1

farmbot-arduino-firmware - latest

roryaronson, 2016-12-14:
Hi @Daniil145, the support for the second X motor will be available in the next release.

Daniil145, 2016-12-14:
[image] 
I Use this now)

fluffy, 2016-12-22:
I just ran into this problem too. Any guess on when the version that supports X2 will be ready ?

TimEvWw, 2016-12-22:
After new year. The code is finished, but needs to be tested before
releasing.

fluffy, 2016-12-22:
I created a patch to drive the two motors and tested it out and sent a PR for it. Seems to be all working for me. Not sure if it meets coding style etc but have a look at it and glad to edit to bring it in line with whatever want.

TimEvWw, 2017-01-26:
The version with X2 is released. That version includes enabling the motor with EEPROM parameters.

closed 2017-01-26 by TimEvWw