FarmBot/farmbot-arduino-firmware #43: the calibration?

Daniil145, 2016-12-16:
Could you tell us what process takes place in the calibration?

TimEvWw, 2016-12-16:
The motor for the choses axis moves to one direction until it reaches an
end point or the decoders detect the motor is slipping. Then it moves in
the other direction, again until it reaches an end point or the decoders
detect the motor is slipping.The amount of steps needed to move between the
two ends is reported back.

Daniil145, 2016-12-17:
is there a code to install encoders and endpoints? If yes, how to connect the endpoints to RAMPS. I have seen how to bridge the encoders. I am a bit do not understand this point. But I saw in the Andes endpoint interface encoders

TimEvWw, 2016-12-18:
The code for encoders is included, but for now disabled by default. You can
plug the encoders in at the pins below, and enable them with the following
commands.

Pin numbering:

16 = X axis encoder A channel
17 = X axis encoder B channel
23 = Y axis encoder A channel
25 = Y axis encoder B channel
27 = Z axis encoder A channel
29 = Z axis encoder B channel

Parameter numbers:

ENCODER_ENABLED_X 101
ENCODER_ENABLED_Y 102
ENCODER_ENABLED_Y 103

Commands to enable encoders (permanently)

F22 P101 V1
F22 P102 V1
F22 P103 V1

bdudout, 2017-02-22:
Hello,

I have some issues during the calibration from the web app. 
Everytime I hit calibration the farmbot moves maximum 1 mm and stops. I use 1000 lines/revolution encoders, powered 5V and I checked with a voltmeter that A and B value are indeed changing from 0 to 5V when I disable encoders and move the farmbot from the controls.

Tested on my.farmbot.io with farmbot os 3.0.5.

With the stable version 1.0 of arduino I get the following log :

Farmbot got a partial gcode: R81 XA0 XB0 YA0 YB0 ZA0 ZB0

When updating the arduino firmware through the web app and trying to calibrate I get the following codes :

Y calibration
unhandled gcode R23 P142 V221 
unhangled gcode R23 P22 V0

Z calibration
unhandled gcode R23 P143 V226 
unhandled gcode R23 P23 V0

Do you have any idea how I can fix that ?

I still didn't tested through serial port but this is my next step!

bdudout, 2017-02-24:
Thanks for replying :)

Yes, when disabling the encoders the motors movements are perfectly fine.
I tested my encoders with a simple arduino interrupt program and the encoders are working fine. My encoders are 1000 lines/revolution so I guess it's making the difference. 

Now I'm testing directly with serial connection the farmbot arduino firmware. It seems there is a polling algorithm used with a timer called every 100 microseconds . I'll test more the firmware and find a way to let my encoders work during the calibration, it seems the function checkAxisVsEncoder() is the right place to figure out what's happening.

TimEvWw, 2017-02-24:
The firmware expects a 100 steps per revolution to match the motors and
monitors to check if the theoretical position and measured position don't
deviate too much. So I expect a 1000 steps encoder to actually cause quite
a panic in the firmware. There is no downscaling build into the firmware
for this.

bdudout, 2017-02-24:
Indeed I realized that my encoders are not matching the current firmware code. I saw before on the forum that any differential encoder of at least 200/lines per revolution would be fine (http://genesis.farmbot.io/discuss/578c96d3bd223d2000cc14bf).

I'll try to modify the firmware to support my 1000 lines encoder. Do you have suggestions on which values I would have to update to get it working ? For example increasing the timer speed to match the encoder, setting MOVEMENT_INTERRUPT_SPEED = 10; 

I already spent days reading and testing the source code but I'm still not an encoder expert :)

TimEvWw, 2017-02-24:
> 578c96d3bd223d2000cc14bf).

In StepperControlEncoder.cpp, function currentPosition, you can replace
"return position;" by "return position / 5;". That way, the encoder
internal position is stays the same, and the rest of the firmware works
with a downscaled position.

bdudout, 2017-02-25:
Great ! I'll try that :) Thanks

roryaronson, 2017-05-06:
The latest firmware has much improved calibration and encoder support. To answer the original question of this issue: simply press the "Calibrate <axis>" buttons on the Device page. Note: you must have Encoders enabled with physical hardstops or have endstops installed.

closed 2017-05-06 by roryaronson