FarmBot/farmbot-arduino-firmware #47: Bug reported by @fluffy on the forum

RickCarlino, 2017-01-02:
I saw that @fluffy reported this bug on the forum, but I thought this might be better discussed as a bug report rather than a general discussion. Here's the text of the post:

 > I went to https://my.farmbot.io/app/devices1 and changed the steps/mm from 500 to 2500. This seemed to send the system into and endless loops of bad GCodes. I just get a stead stream of errors on the the place on web where errors seem to go.

 > Any thoughts on what went wrong or ways to test it and log proper bugs?

 > An example of the errors look like

```json
{
"id": 29092,
"message": "Farmbot got an unhandled gcode! {:unhandled_gcode, \"Q0\"}",
"meta": {
"x": -1,
"y": -1,
"z": -1,
"type": "info"
},
"channels": [],
"device_id": 250,
"created_at": "2016-12-27T23:58:58.968Z",
"updated_at": "2016-12-27T23:58:58.968Z"
},
{
"id": 29093,
"message": "Farmbot got an unhandled gcode! {:unhandled_gcode, \"Command: F21 P62d wh ode: : : : : V W:: : : : ecmd Q:2\"}",
"meta": {
"x": -1,
"y": -1,
"z": -1,
"type": "info"
},
"channels": [],
"device_id": 250,
"created_at": "2016-12-27T23:58:58.972Z",
"updated_at": "2016-12-27T23:58:58.972Z"
},
{
"id": 29094,
"message": "Farmbot got an unhandled gcode! {:unhandled_gcode, \"denu - \n0112and:21 62 Q0\r,\"}",
"meta": {
"x": -1,
"y": -1,
"z": -1,
"type": "info"
},
"channels": [],
"device_id": 250,
"created_at": "2016-12-27T23:58:58.976Z",
"updated_at": "2016-12-27T23:58:58.976Z"
},
{
"id": 29095,
"message": "Farmbot got a message from my arduino: Command with code: 121, X: 0.00, Y: 0.00, Z: 0.00, S: 0.00, P: 62, V: 0, W: 0, T: 0, E: 0, M: 0, Q: 0",
"meta": {
"x": -1,
"y": -1,
"z": -1,
"type": "info"
},
"channels": [],
"device_id": 250,
"created_at": "2016-12-27T23:58:58.982Z",
"updated_at": "2016-12-27T23:58:58.982Z"
},
{
"id": 29096,
"message": "Farmbot got an unhandled gcode! {:unhandled_gcode, \"exec cmd Q: 0\"}",
"meta": {
"x": -1,
"y": -1,
"z": -1,
"type": "info"
},
"channels": [],
"device_id": 250,
"created_at": "2016-12-27T23:58:58.989Z",
"updated_at": "2016-12-27T23:58:58.989Z"
},
{
"id": 29097,
"message": "Farmbot got an unhandled gcode! {:unhandled_gcode, \"c2\"}",
"meta": {
"x": -1,
"y": -1,
"z": -1,
"type": "info"
},
"channels": [],
"device_id": 250,
"created_at": "2016-12-27T23:58:58.992Z",
"updated_at": "2016-12-27T23:58:58.992Z"
},
{
"id": 29098,
"message": "Farmbot got an unhandled gcode! {:unhandled_gcode, \"c3: 121\"}",
"meta": {
"x": -1,
"y": -1,
"z": -1,
"type": "info"
},
"channels": [],
"device_id": 250,
"created_at": "2016-12-27T23:58:58.996Z",
"updated_at": "2016-12-27T23:58:58.996Z"
},
{
"id": 29099,
"message": "Farmbot got an unhandled gcode! {:unhandled_gcode, \"code Enum C - 121\"}",
"meta": {
"x": -1,
"y": -1,
"z": -1,
"type": "info"
},
"channels": [],
"device_id": 250,
"created_at": "2016-12-27T23:58:59.000Z",
"updated_at": "2016-12-27T23:58:59.000Z"
},
{
"id": 29100,
"message": "Farmbot got an unhandled gcode! {:unhandled_gcode, \"Q0\"}",
"meta": {
"x": -1,
"y": -1,
"z": -1,
"type": "info"
},
"channels": [],
"device_id": 250,
"created_at": "2016-12-27T23:58:59.003Z",
"updated_at": "2016-12-27T23:58:59.003Z"
},
{
"id": 29101,
"message": "Farmbot got an unhandled gcode! {:unhandled_gcode, \"Command: F21 P63 Q0\r,\"}",
"meta": {
"x": -1,
"y": -1,
"z": -1,
"type": "info"
},
"channels": [],
"device_id": 250,
"created_at": "2016-12-27T23:58:59.006Z",
"updated_at": "2016-12-27T23:58:59.006Z"
},
{
"id": 29102,
"message": "Farmbot got a message from my arduino: Command with code: 121, X: 0.00, Y: 0.00, Z: 0.00, S: 0.00, P: 63, V: 0, W: 0, T: 0, E: 0, M: 0, Q: 0",
"meta": {
"x": -1,
"y": -1,
"z": -1,
"type": "info"
},
"channels": [],
"device_id": 250,
"created_at": "2016-12-27T23:58:59.010Z",
"updated_at": "2016-12-27T23:58:59.010Z"
},
{
"id": 29103,
"message": "Farmbot got an unhandled gcode! {:unhandled_gcode, \"exec cmd Q: 0\"}",
"meta": {
"x": -1,
"y": -1,
"z": -1,
"type": "info"
},
"channels": [],
"device_id": 250,
"created_at": "2016-12-27T23:58:59.014Z",
"updated_at": "2016-12-27T23:58:59.014Z"
}
```

CC: @ConnorRigby

ConnorRigby, 2017-01-02:
@fluffy What version of the Firmware and OS are you running? it looks to me like your arduino and bot got a bit out of sync. i would give a reboot to both, and see if this happens again. As to the endless loop of G codes, i dont know what caused it but i dont suspect it as an endless loop, but more of a problem where neither devices can communicate to eachother because the codes arent in the buffer where they are expected.

fluffy, 2017-01-18:
That could be because my Arduino build is the curren head of master branch plus the patches in the PRs I have sent over to main repo to do things like support support 2 x axis motors and such. Is there a git tag or branch for the latest version of the Arduino code I should be running? I think I had the latest Pi code.

RickCarlino, 2017-01-18:
@fluffy FarmBot OS is tested on (and is runs) firmware version `2.0.0`, which is the latest version on the [releases page](https://github.com/FarmBot/farmbot-arduino-firmware/releases). Your theory sounds correct, as we do not develop FarmBot OS on edge versions of the firmware. For Pi code, we just released [FarmBot OS 3.0.1](https://github.com/FarmBot/farmbot_os/releases). If you want to try 3.0.1, I would recommend using the [staging server](http://staging.farmbot.io) until we do a production release (hopefully today or tomorrow).

Hope that helps. Please let us know if you need any other help.

closed 2017-01-18 by RickCarlino