FarmBot/farmbot-arduino-firmware #54: Motor X2 not working

AscendFB, 2017-02-03:
Hello everyone,

my second motor on X-Axis is still not working at all. Using the newest firmware.
What I tried so far:
- update via staging.farmbot.io
- upload directly to arduino
- versions a few commits earlier

Uploading the dual motor version of fluffy helps out, but i think it may be outdated since its from mid December. 

Is there anything needed to be done manually to get the X2 motor running? Or is it just activated by default?

Also there are sometimes messages from arduino, just like this:
[image]
but they appear for every axis.

TimEvWw, 2017-02-03:
Can confirm there is a bug for X2. Working on it.

vaibhavgupta3110, 2017-02-06:
Added the E parameters to the farmbot_arduino_controller.cpp file and it's confirmed working perfectly, you can test out the config from this PR

#55

roryaronson, 2017-02-06:
Hooray!

closed 2017-02-06 by roryaronson