FarmBot/farmbot-arduino-firmware #58: 3.0.3 Version Problem

Daniil145, 2017-02-17:
Hi i install new release and bot connect and disconect every 1 sec to api(log devise online, offline)
Previos version work normaly but more unhandled errors in logs

TimEvWw, 2017-02-17:
Does the error show on the web page or are you testing directly with the
serial communication?

Daniil145, 2017-02-19:
I use web interface or api. I see errors on web interface.
The previous version worked fine but more unhandled errors in logs. This version is constantly Disconnects and re-connect again every sec
I use last release version farmbot_os.

RickCarlino, 2017-02-20:
@Daniil145 Which server are you on? `my.farmbot.io` or `staging.farmbot.io`? Also, could you copy/paste the errors so I can see them. Thanks.

bdudout, 2017-02-22:
I had the same issues with Daniil145, before there was a lot of unhandled messages and now I sometimes have the disconnect/connect issue, 1 second up, 1 second down. 

Tested on my.farmbot.io with farmbot os 3.0.5 with latest version of farmbot arduino firmware compiled with arduino 1.8.1

Here is the log I can get :
FARMBOT IS UP AND RUNNING!
2:38PM
device_484 is offline!
1:15pm
Farmbot is up and running!
2:38pm
device_484 is offline!
1:15pm

etc, goes like this forever

This might be due to the latest version of the arduino firmware because I do not have this problem with the stable 1.0 version.

closed 2017-04-26 by roryaronson