FarmBot/farmbot-arduino-firmware #62: Dont work endpoint after update

Daniil145, 2017-03-16:
I use whis version
[image]

RickCarlino, 2017-03-16:
@Daniil145 It looks like you're running an old version of the frontend. Firmware and FBOS are no longer updated separately.

Please update your API and run `./install_frontend.sh`.

Also, make sure that you run `ctrl + shift + r` to clear your browser cache (in case your browser has an old version in cache).

It should look like this:

[image]

closed 2017-03-16 by RickCarlino

Daniil145, 2017-03-16:
Endpoints work on new version?

Daniil145, 2017-03-16:
[image]
I updated and test it tommorow