FarmBot/farmbot-arduino-firmware #65: z-axis is moving due to its own weight

bautze, 2017-04-12:
If z-axis is running smooth using high quality grease, the z-axis moves down due to gravity. 

The stepper motors usually (CNC, 3d-printer) can be disabled and enabled. Have a look at http://reprap.org/wiki/G-code
In the disabled state, they can be moved freely, wherever in the enabled state, they are blocking. This seems to be a software issue as the enabled state should be the correct one. Several users are facing the issue in the Farmbot Europe community.

RickCarlino, 2017-04-12:
@bautze We are working on a checkbox in the device settings page to handle this issue. I will keep this issue open and close when a fix is deployed.

bautze, 2017-04-13:
Not sure that another checkbox in the long term is the perfect solution as the default value should be the correct one for most users. An additional setting might confuse unfamiliar users, however I do agree that this setting is useful for most advanced users and should be settable.

roryaronson, 2017-05-06:
@bautze toggle switches have been added to keep each axis locked or unlocked. The default setting for the z-axis is to remain locked.

closed 2017-05-06 by roryaronson