FarmBot/farmbot-arduino-firmware #98: the stepper moved too slow?

yunfan, 2017-11-25:
hi my friend bought a farmbot genesis, and found it moved too slow, well, if we replace the builtin arduino board to another one, they will become extreming fast, but lost acurracy.

currently i tried connect to that arduino directly with my pc and send gcode and other commands via serial console. the problem is they just act too slow while using G00 and G01. 

i tried to add a duplicate `PORTF |= B00000001;`   in `StepperControlAxis::setMotorStepWrite54`  and it made the z axis moved faster twice, but it might not the a right solution, just need some technical expanation

AscendFB, 2017-11-26:
Hey 
did you check if there is microstepping active (jumpers below the stepper drivers)?
I'm just asking because if its active, steps are beeing divided and a lower speed is the result.
When you replaced the arduino (with the ramps I guess) you used one which had no jumpers connected.

yunfan, 2017-11-28:
@AscendFB thanks for the information, i'll ask them to check that. also i notice that the newest firmware's version is 5.x , but it looks this repo only have 4.x code, so what were you guys updated? when will you release the code?

yunfan, 2017-11-28:
@AscendFB   also confirmed that our farmbot dont have divided enable

ConnorRigby, 2019-09-19:
Closing due to inactivity.

closed 2019-09-19 by ConnorRigby