FarmBot/farmbot-arduino-firmware #99: Wont compile

kezzab25, 2018-02-26:
It had a few issues when I first compiled. One was defining NULL and others were something about long ints? I had to change a few 0.0 to 0 to make it compile. Why was this?
I managed to make a few changes and now it compiles but I cant remember what I changed.

gabrielburnworth, 2018-02-26:
Do you have Mega 2560 selected as the board to compile for? What version of the Arduino IDE are you using?

kezzab25, 2018-02-26:
I am using 1.8.1 and yes I have mega 2560 selected. It uploads fine once I
fix the errors. I'm just wondering why I'm getting errors in the first
place?

gabrielburnworth, 2018-02-26:
I am using `Arduino 1.8.1` and `Mega 2560` and I don't get any compile errors on `3ad1d88`.

You can try:
```
git clone -b 'v6.0.1' --single-branch  https://github.com/FarmBot/farmbot-arduino-firmware
```

kezzab25, 2018-02-26:
[image]

kezzab25, 2018-02-26:
Sorry, the one that wont compile is version 1.8.4

gabrielburnworth, 2018-02-26:
Ah, those are complier warnings, not errors. The build should still compile successfully. Do you not get `Done compiling.`?

kezzab25, 2018-02-26:
I was a bit impatient. It does eventually upload if I wait a bit longer

closed 2018-02-26 by kezzab25