FarmBot/farmbot-arduino-firmware #135: [bug, minor] Firmware interface reports wrong dv/dt for Z-axis state

jsimmonds2, 2020-02-24:
Playing with the `FarmbotFirmware` interface to the Farmduino and noticed that the **accelerate** and **decelerate** reports for **Z**-axis state are "upside-down".

Picture this :-)

[image]

gabrielburnworth, 2020-02-24:
This may occur on any axis when moving in the negative direction:

https://github.com/FarmBot/farmbot-arduino-firmware/blob/ffe3573be542226ae68fd7ec84cdbc3011bf0895/src/MovementAxis.cpp#L204-L239

jsimmonds2, 2020-02-25:
Oh, Ok.  Not a "mathematical" state report, but a funky FarmBot thing ?

From [Merriam Webster](https://www.merriam-webster.com/dictionary/acceleration) :
`Definition of acceleration
1a: the act or process of moving faster or happening more quickly`