FarmBot/farmbot-arduino-firmware #156: CommitSHA.h incorrectly formatted

jsimmonds2, 2020-09-25:
Used Arduino IDE 1.8.13 to **make** the current farmbot-arduino-firmware version and compilation failed because of
```c++
src/CommitSHA.h
```
[image]

jsimmonds2, 2020-09-25:
Ooops, sorry, my environment is Fedora Linux 64bit version 32 Workstation
```
Linux f32v-on-comet 5.8.10-200.fc32.x86_64 #1 SMP Thu Sep 17 16:48:25 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```
I'll try to workout why **Makefile** doesn't create a useable **CommitSHA.h** on this platform.

jsimmonds2, 2020-09-26:
Found it.  I'll send a PR for Makefile.

closed 2020-09-26 by gabrielburnworth

gabrielburnworth, 2020-09-26:
Thanks for identifying and fixing the issue in your environment! As you mentioned, the previous code has been working on CI (and our systems) because it was using the builtin `echo`, which didn't need the `-e` option. Also, as you may have noticed on CI, adding the `-e` option didn't work with the builtin so I changed it to `/bin/echo` to avoid environment confusion.

jsimmonds2, 2020-09-27:
@gabrielburnworth thanks so much for sorting out a better solution ❤️ 

[ For my own learning, . .  Which Linux and which Shell are used in CircleCI ? ]

F.y.i. While researching my solution I discovered that "best-practice" Makefiles define a SHELL near the beginning 🙂

gabrielburnworth, 2020-09-27:
[CircleCI uses Ubuntu](https://circleci.com/docs/2.0/configuration-reference/#available-machine-images).

Thanks for the tip. (Reference docs: [choosing the shell](https://www.gnu.org/software/make/manual/html_node/Choosing-the-Shell.html), [conventions](https://www.gnu.org/software/make/manual/html_node/Makefile-Basics.html))

I've added: https://github.com/FarmBot/farmbot-arduino-firmware/blob/099bb705df9c6f8ebbac5f502cd27539a8a5b16e/Makefile#L1