FarmBot/farmbot-arduino-firmware #161: [FEATURE REQUEST] Repository for Farmbot PCBs

darigovresearch, 2021-04-18:
As there aren't any repositories for PCB design thought it would be best to open an issue in the repository which is closest.

Having version-controlled based repository may help certain users which want to test and build the PCBs in order to improve or adapt them to weird edge cases to have a repository specifically for PCBs which contain the source files (EAGLE, KiCad etc) and the Gerbers. That way users can use tools like [kitspace](https://kitspace.org/) to easily order components based on the BOM and Gerbers.

Existing CAD models can be found here but think that having a repository for will improve the update and development of PCB based designs from the community and may simplify your workload as well - https://genesis.farm.bot/v1.5/Extras/cad

Example repositories which do this well
Open source e-reader - [here](http://github.com/joeycastillo/The-Open-Book)
Open source filament diameter checker - [here](https://github.com/drspangle/infidel-sensor/)

Would love to get your thoughts on this and would be happy to generate the Kitspace files if given Gerbers and an electronics BOM if any of the component suppliers you use are integrated into Kitspace.

darigovresearch, 2021-05-03:
@RickCarlino any thoughts on the above?

Happy to discuss if that helps

RickCarlino, 2021-05-03:
@darigovresearch No interest on our end, but if you wish to maintain an independent mirror on Github and link to it here you may do so.

closed 2021-05-03 by RickCarlino

darigovresearch, 2021-05-03:
Thanks for the clarification!