FarmBot/farmbot-arduino-firmware #163: How to configure the code if i replace the RAMPs 1.4 with Arduino Mega by MKS GEN 1.4 ?

alkhlama, 2021-12-08:


RickCarlino, 2021-12-08:
@alkhlama This is a very difficult task and it would be easier to just use a supported board, such as RAMPS. I have never worked with the MKS GEN 1.4 before and do not have plans to add support for it, since our main focus is supporting users of official kits.

If you wanted to do this, you would need to:

 * Write a GCode interpreter for the target platform that is compatible with the FarmBot firmware GCode spec (see README.md)
 * Create a `*.hex` file for the firmware.
 * Put the firmware blob onto the SD card as `custom.hex`.
 * Change the device firmware path in the advanced settings menu (`/dev/ttyUSB0`, `/dev/ttyACM0`, etc...)
 * Reboot the device.
 
It's important to note that adding support for arbitrary controller boards is possible, but it is not a goal of the project.

closed 2021-12-08 by RickCarlino

RickCarlino, 2021-12-08:
Quick update: I was not aware that the MKS v1.4 is RAMPS compatible. If that is the case, you most likely need to just change your "FIRMWARE PATH" after enabling "ADVANCED SETTINGS":

[image]
[image]

alkhlama, 2021-12-09:
Thank you a lot it work now after I change the "FIRMWARE PATH"  
I highly appreciated you support

alkhlama, 2021-12-11:
Greetings
Can I change the farmbot from Cartesian to Polar Representation and how we can increase the speed of the robot
Your support is highly appreciated

[image]<https://user-images.githubusercontent.com/1388608/145262534-6ca8ae31-0cf7-42d1-94d3-243c774a9589.png>
[image]<https://user-images.githubusercontent.com/1388608/145262444-84c44e79-d44a-4051-b1c0-0c1b4055bb77.png>

RickCarlino, 2021-12-11:
@alkhlama For general discussion, please see the [FarmBot forum](https://forum.farmbot.org/). We use Github for the discussion of specific developer-focused software problems and the main issue in this thread has already been solved/closed.

> Can I change the farmbot from Cartesian to Polar Representation

You would need to change most aspects of the FarmBot software stack (firmware, FBOS, Web App, possibly API).

Some people have attempted this as a research project: https://digitalcommons.calpoly.edu/imesp/205/

I am not aware of any working, production-ready polar FarmBot systems in operation.

 > how we can increase the speed of the robot
 
You can increase the step count, but this comes at the cost of increasing the likelihood of a stall. Higher speeds mean you need straighter rails that are less tolerant of variation. If you notice that the device's movements are no longer reliable, that may be a sign that you set the speed too high.

Please see the [official documentation](https://software.farm.bot/v14/The-FarmBot-Web-App/settings/motors.html#steps-per-mm) for more information.